michar71/Open-DSO-150 #2: Encoder/Button control problematic

michar71, 2017-04-04:
Encoder and button performance on DSO-150 is pretty bad. The underlying issue is the dual use of GPIOB for display data output and encoder/button input.

Toggling IRQ's on/off when display is using the lines kind of works. (And could be improved by linking it directly to the CS line enable/disable by the display driver). 
It is also possible to try to just set a flag in the display driver code when accessing the data lines and just branching out of the IRQ routine while the display is using the lines.

None of this though will prevent user interactions to get lost. 

For the encoder this is not that problematic as the user just continues to operate the encoder until the desired action happens. (Reducing the dead-time will improve response...)

For buttons this is a bigger problem. If the IRQ is disabled while the button is pressed/released the event is lost. This leads to either lost key events or key presses getting registered as long presses as the short release pulse is missed.

Therefore currently buttons are polled which allows the level of the line to be monitored at the polling time and not rely on edge transitions but this causes issues when the timebase gets longer as the aquisition routine blocks polling. Switching to DMA for the data aquisition might improve this.

jief666, 2017-04-04:
On my current DSO 150 with JYE firmware, the encoder and button works quite
well. I think that happens to lose some click but very rarely.
I don't know the average time a button stay down when you press it, but if
it's longer than a screen access, we might catch them all...

michar71, 2017-04-04:
Buttons work fine (With polling...) right now.
But via IRQ if we miss the falling/rising edge because the IRQ is currently disabled due to Display access the event is gone. 
But polling gets worse with longer time delays in the sampling routine when it starts to take longer. DMA transfer is the solution for that because it won't block the polling.