michar71/Open-DSO-150 #3: Implement DMA driven Data Sampling Routine

michar71, 2017-04-04:
Currently data acquisition is done by polling the ADC's which will block buttons and limits the maximum sampling rate. A lot of those limitations could be resolved by using DMA driven ADC data collection.
Digital Input should still be possible by using ADC IRQ to trigger digital line storage.
Acquisition mode could probably be improved by using continuos mode into  a buffer and then use the trigger to stop ADC capture after 1/2 buffer size (So we have data from before the trigger in the Buffer...)

ADC StartConversion() Routine
1)	Setup ADC Analog Settings/Capture Mode
2)	Setup Timer1 as external ADC Trigger Source
3)	Setup Timer2 as Timeout Timer
4)	Link Timer2 IRQ to Trigger IRQ Routine
5)	Setup ADC Analog Settings/Capture Mode
6)	Setup DMA Address/Conversion Length
7)	Set Start Address for Digital Input Capture Array
8)	Link ADC Conversion Complete IRQ to Input Capture IRQ Routine
9)	Link DMA complete IRQ to DMA IRQ Routine

Trigger IRQ Routine
1)	Disable Trigger IRQ
2)	Start ADC External Trigger Timer

Input Capture IRQ Routine
1)	Store IO Register in Array
2)	Increase Array Pointer

DMA IRQ Routine
1)	Disable Input Capture IRQ
2)	Disable Timer1

universam1, 2017-06-16:
Here https://github.com/pingumacpenguin/STM32-O-Scope/blob/master/STM32-O-Scope.ino#L492 is an example of implementation, hope that is applicable?

michar71, 2018-03-31:
After looking into this DMA capture has numbers drawbacks (Can't capture GPIO's, not much faster, can't abort, buttons no longer an issue with IRQ's...)

closed 2018-03-31 by michar71