michar71/Open-DSO-150 #5: Creating GUI for displaying data

nemanjan00, 2017-05-10:
Hello, from what I see, you have very similar code to DLO-138, when it comes to dumping samples to Serial... 

I am writing GUI for displaying data from DLO-138. 

Would you be willing to test it with 150? 

I do not have hardware... 

For now, it is just CLI, parsing output... 

If it does parse, then, it will work with GUI also...

michar71, 2017-06-20:
We can work on that. I currently have a menu item "Dump Serial Data" but it might be better to also have an option to continuously dump samples in binary format to get more data throughput.
Do you already have a format for the data?

nemanjan00, 2017-06-22:
So far, I did write parser for existing dump... 
And found couple of bugs while doing it.

Harvie, 2019-11-25:
I don't think it's worth the effort to write GUI specific for single device.
There is sigrok/pulseview project, which aims to support lots of devices https://sigrok.org/wiki/Supported_hardware

There are two ways to add support:

1.) take serial protocol already supported by sigrok and implement it in firmware
2.) write own sigrok driver to support custom protocol used in firmware and make pull request so it can be merged by sigrok community