michar71/Open-DSO-150 #13: bin file

tayfunulu, 2017-09-08:
hi everyone, is it possible to upload a bin file for dso150 ?

namadori, 2017-12-21:
That would be nice

sreichholf, 2018-04-17:
You could give this one a try.
Keep OK pressed on first powerup (press it before powering) to make sure your config reset properly.

[dso_150.zip](https://github.com/michar71/Open-DSO-150/files/1918185/dso_150.zip)

daividtux12, 2018-04-24:
Hi! good evening! How to generate the hexadecimal file for the JYE DSO150 using Atollic TrueSTUDIO?

rtek1000, 2018-05-11:
Hello,

Very good guys!

I'm going to try this firmware,

But I'd like to use 2 analog inputs.

I saw some videos on YouTube, I do not even need another analog card, the signal I want to read is between 0 and 3.3V.

I have an original DSO150, I still have the activation code for the new original firmware, it's a shame that it does not use USB communication for remote control, similar to a HANTEK.

patrickbonner, 2018-05-21:
hi guys.

I think i have just a clone. After flashing Opensource Firmware screen is flashing short "c++ dso 150" in a loop... :-/

could i have a full Flash Dump for reflashing my device? 

best regards
Paddy

HA4ever37, 2018-06-25:
Unfortunately I have one of fake boards and flashing Open-DSO-150 made my screen frozen. However, I was able to make the board work again by flashing  113-15011-60B.zip from this link https://jyetech.com/forum/viewtopic.php?f=19&t=1206

mpovel, 2019-02-16:
At least building HEX file the changes required are:
Go to project->Build Settings->Tool Settings tab->in other->Output format->Tick Convert build output and select desired output file type.

Harvie, 2019-12-04:
> @sreichholf commented on 17 Apr 2018 • dso_150.zip

I've tried this build and it seems to work just perfect! Would be great to have official releases of latest stable build when there's new version.

> Unfortunately I have one of fake boards and flashing Open-DSO-150 made my screen frozen. 

It worked for me on fake board. Original jyetech firmware didn't.

GEEK-in-the-shell, 2019-12-24:
> > @sreichholf commented on 17 Apr 2018 • dso_150.zip
> 
> I've tried this build and it seems to work just perfect! I've flashed it using stm32flash program on linux, through the serial bootloader. Sometimes pressing the jog button does not switch to another feature on display, so i have to press different button to get it working again. But i am not sure if that's a bug or a feature... Maybe i just entered some different operation mode, that i am not aware of (i just went through the README file real fast)
> 
> Would be great to have official releases of latest stable build when there's new version.
> 
> > Unfortunately I have one of fake boards and flashing Open-DSO-150 made my screen frozen.
> 
> It worked for me on fake board. Original jyetech firmware didn't.

Hello Harvie, you can upload the firmware? I have a fake board and the screen is always white :(

Thank you!

rtek1000, 2019-12-24:
Does the official developer not sell valid code for fake boards?

I constantly use the DSO150 for low frequency projects, the original firmware suits me well.  And for slightly larger projects I use the DSO138 (30MHz), this DSO138 is not very good for low frequency, very bad firmware, but the uC is a STM32F407, and uses external ADC.

GEEK-in-the-shell, 2019-12-24:
I tried the firmware of sreichholf but my DSO150 is blinking on the first screen.

Harvie, 2019-12-24:
> Does the official developer not sell valid code for fake boards?

I don't think so, but haven't really checked. Have you seen such offer?

> > > @sreichholf commented on 17 Apr 2018 • dso_150.zip
> Hello Harvie, you can upload the firmware? I have a fake board and the screen is always white :(

Why? It's already here: https://github.com/michar71/Open-DSO-150/issues/13#issuecomment-381842289

rtek1000, 2019-12-24:
> > Does the official developer not sell valid code for fake boards?
> 
> I don't think so, but haven't really checked. Have you seen such offer?

I haven't seen anyone confirm this yet.

But I believe it is possible to contact the developer via email.

His email is available to send the serial number, and then receive the firmware activation code.

I bought the original kit, so I didn't have to ask if it would be possible to buy an activation, but if Microsoft makes money selling software validation, why wouldn't the developer sell an activation?

The cost of the official kit is $39 ~ $43, I don't know how much the activation value can be, but it might be an alternative, for the user to have a working device, and for the developer to have some financial return on development. Maybe it's a question of asking less aggressively, it might help to say that you need firmware activation for a personal parallel board, not a fake one.

E-mail: jyetek@gmail.com (From: https://jyetech.com/firmware-dso-150-shell/)

Store: https://accudiy.com/products/dso-shell-dso150-oscilloscope-diy-kit 

Good luck!

(Note: If anyone gets positive feedback from the developer, please let us know about the cost.)

Harvie, 2019-12-24:
To be honest, the OpenDSO is so nice, i don't feel like i need the original firmware at all. It probably even has more features than original...

GEEK-in-the-shell, 2019-12-25:
> > Does the official developer not sell valid code for fake boards?
> 
> I don't think so, but haven't really checked. Have you seen such offer?
> 
> > > > @sreichholf commented on 17 Apr 2018 • dso_150.zip
> > > > Hello Harvie, you can upload the firmware? I have a fake board and the screen is always white :(
> 
> Why? It's already here: [#13 (comment)](https://github.com/michar71/Open-DSO-150/issues/13#issuecomment-381842289)

I try it but not work for me, blinking on the first screen.

What program do you use to load the firmware?

Harvie, 2019-12-25:
> What program do you use to load the firmware?

stm32flash

GEEK-in-the-shell, 2019-12-25:
> > What program do you use to load the firmware?
> 
> stm32flash (mentioned in my first comment)

My dso150 works again!! , it hurts not being able to use openDSO but only works with 113-15011-60B..

Thank you guys!

josla972, 2020-04-05:
I have a fake board, but by skipping the EE_Reads call in zconfig.cpp and rebuilding I got this software running.

@@ -50,7 +68,7 @@ void loadConfig(bool reset)
 
 
        // read preamble
-       if(reset)
+       if(1)//(reset)

Btw, I also made an experimental interrupt based frequency calculation fix since I was not really pleased with the current performance if somebody is interested.

universam1, 2020-05-09:
@michar71 Would you be in favor to add PlatformIO build and flash support?
I consider it the best option these days, greatly simplifying CI/CD and being totally portable.

Created a working MR https://github.com/michar71/Open-DSO-150/pull/43
flashed my personal DSO-150 also with it fine!

michar71, 2020-05-09:
Hi,

Funny, I just started a complete rewrite of the whole code after I looked at the code after 2 years and seeing how ugly things are.
It’s an exercise in software architecture (that is turning quickly into an exercise for a small footprint graphics framework ;-) so the UI can be decoupled from the underlying scope parts...
Also adding cpu overclocking, DMA and a bunch of other things that would be a total hack in the old Codebase.

It‘s interesting that you mention platformIO. I looked into that a while ago for stm32 and had a bunch of issues with debugging. 
For the new codebase Im using STM32Ide mostly because that‘s what (I have to) use at work... But I‘ll give platformIO another try after I have gotten at least some basic code to compile/run...

Michael

rtek1000, 2020-05-09:
Correction of my previous commentDSO338 not DSO138:

Ref.: https://lygte-info.dk/review/Equipment%20Oscilloscope%20Aarontek%20DSO338%20UK.html

universam1, 2020-05-10:
Hi @michar71 
Changing the project to STM32ide sounds like a good idea to me, as I struggled and finally gave up to with Atollic to built the project.

From my https://github.com/michar71/Open-DSO-150/pull/43 you can see that they need not to be exclusive but PIO can be supplementary to whatever IDE you choose, just as a simple compiling solution for end-users that do not necessarily like to install the full blown IDE just to get the binary.

Actually, using Travis this would allow to set up automatic pipeline binary release on every commit or tag as you like, have great experience in my projects

driehoek91, 2020-05-20:
I tried flashing with v120 on my fake DSO150 and voila came out a fake board message . after that I flash with Open DSO150 the screen just blinked white, when I tried to read the log via UART it turned out that the cpu (STM32) kept rebooting repeatedly, it turns out that the v120 firmware disabled the watchdog software (WDG_SW) in cpu.

I just needed to enable WDG_SW with UART (with "Flash Loader Demonstrator v2.7.0", if i use the newer version there is no "option bytes" menu to change WDG_SW). or i can use ST-Link v2 dongle (with "STM32 ST-Link Utility")

focus77, 2020-06-02:
Alla prima accensione, tenere premuto il tasto ok

semvis123, 2020-06-16:
Is there a recent hex version to download? I couldn't get the firmware to compile

jairsinho, 2020-08-02:
> I tried flashing with v120 on my fake DSO150 and voila came out a fake board message . after that I flash with Open DSO150 the screen just blinked white, when I tried to read the log via UART it [...]

Have you tried to wire 3.3V to RESET and then reflash the v64?

[image]

gregcrago, 2020-09-05:
I gave up on Atollic TrueSTUDIO, impossible to build code. Went with STM32CubeIDE v1.4.0, imported Atollic TrueSTUDIO project, built in 17 sec. Used SMT32 ST-LINK to Flash Bin file found in new \DEBUG folder under project folder. Checksum for ver 3.3 is 0x004f0932, Make sure to hold OK while powering up to reset config. Has to AUTOCAL TWICE to get A1 signal to ground. Using 'real' JYE Tech board.

gregcrago, 2020-09-05:
Used ST-Link V2 dongle, Could not get Bootloader to run by jumping JP1 and JP2. When you use ST-Link, do use DIO pins and do not need to jumper/desolder the 2 pins anymore, Much easier. Solder Female 90 header and cut slot in case to allow easy flashing in future.

jackthese, 2021-04-14:
jairshinho... relative to 

> Have you tried to wire 3.3V to RESET and then reflash the v64?

I connected the 3.3v to reset, turned it on, and it started up normally with no fake board message. I did not have to reflash. I soldered that jumper in permanently, and the unit seems to work normally. I'm just a dumb old mechanical engineer with little experience using a 'scope, but I'm seeing my ESP32 PWM wave forms just fine. Thanks for the tip...

jairsinho, 2021-04-15:
> jairshinho... relative to
>
>
> I connected the 3.3v to reset, turned it on, and it started up normally with no fake board message. I did not have to reflash. I soldered that jumper in permanently, and [...]

Jack,

glad to hear it worked for you 👍 

"You are never too old or too wise to learn"

loglow, 2021-07-29:
Well...

It took me way too much work to figure everything out, but I got the current code (v3.3) compiled in TrueSTUDIO v9.3.0 today. The hex file really should be published as a download here somewhere, because that took forever and I had to make a whole new project from scratch since the software changed so much, and I had to dig through the settings files manually to get all the symbols and flags in there correctly.

This is the compiled binary (hex) for flashing, zipped up for GitHub to allow it. I used a Segger J-Link to flash it via SWD and it was super easy. The chip is an STM32F103C8.
[Open-DSO-150-v3.3.hex.zip](https://github.com/michar71/Open-DSO-150/files/6898555/Open-DSO-150-v3.3.hex.zip)

This is the source and project folder that I used to build it, including the hidden ".settings" directory and its contents.
[Open-DSO-150-v3.3-TrueSTUDIO-v9.3.0.zip](https://github.com/michar71/Open-DSO-150/files/6898558/Open-DSO-150-v3.3-TrueSTUDIO-v9.3.0.zip)