michar71/Open-DSO-150 #29: Just got a clone...  Should I load this firmware?

jbuszkie, 2019-04-19:
I just got a clone of the DSO 150.   I didn't do my home work before I purchased it! :-(
With the current firmware (V62)  I can't display correctly @ 20us and 10us/div.
I can see it triggered, but I get no waveform on the display.
I was going to try this open DSO 150 thing but I can't seem to save the original firmware.
It seems to be protected. I have to erase everything.  So...
Will this open DSO fix my triggering issues?
Is there a pre-compiled hex to load?
Is there anyway to go back to the clone V62 firmware if the open DSO doesn't work for me?

Thanks,

Jim

Mjolinor, 2019-04-19:
I just loaded this onto my £14 clone one off Aliexpress with no problems.

That is not saying yours will be OK, just that mine was. I had the same reservation but at £14 took the risk and it worked out fine. I tried taking a backup before but could not read successfully so just wrote with hope. :)

jbuszkie, 2019-04-21:
I just got this to work!  Thanks!  This firmware is really cool!

closed 2019-04-22 by jbuszkie