michar71/Open-DSO-150 #30: How to build....

jbuszkie, 2019-04-19:
Ok..  I loaded the atollic true studio thing.  
I downloaded this project.
I imported  the project.
I went to build project and it take two seconds with no bin or hex generation.  I really think it's not building everything!  Here is the console output:
12:04:34 **** Rebuild of configuration Debug for project dso_150 ****
Info: Internal Builder is used for build
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m3 -std=gnu11 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F103xB -I../Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F1xx/Include -I../Drivers/CMSIS/Include -O1 -ffunction-sections -fdata-sections -fno-strict-aliasing -g -fstack-usage -Wall -o Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_gpio_ex.o ..\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_gpio_ex.c 
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m3 -std=gnu11 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F103xB -I../Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F1xx/Include -I../Drivers/CMSIS/Include -O1 -ffunction-sections -fdata-sections -fno-strict-aliasing -g -fstack-usage -Wall -o Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_dma.o ..\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_dma.c 
Info: Parallel threads used: 2

12:04:34 Build Finished (took 485ms)

I'm sure I'm missing something stupid...  this is version 9.3.0 of the truestudio.

Mjolinor, 2019-04-19:
Follow this and select hex rather than bin.
https://community.st.com/s/question/0D50X00009XkWfzSAF/how-to-generate-a-bin-file-in-atollic

jbuszkie, 2019-04-19:
I don't even have a bin file.  It should take longer than 485ms to build, right?
It only seems like it's building the drivers folder? 
And I did see where to output a hex vs bin..  But I don't even have a bin yet.
I tried several earlier versions of the tool and I still get the same results.  Super quick build but no bin/hex.

The out above was ALL the output.  not just a snippet..

Thanks for the help so far! :-)

I'm sure it's just something stupid I'm doing...  or not doing.  Do I have to set any settings.  It sounded like I just download the project from git hub and load it into the tool and hit build project, right?

Jim

jbuszkie, 2019-04-21:
I'm building on an old WinXP machine?  Should that matter?  I tried injecting errors into main.cpp by changing a header file name to crap and the build didn't complain.  So I really don't think it's trying to compile main.  Any Suggestions?

jbuszkie, 2019-04-21:
It was a Windows XP thing.  I installed TrueStudio on a Win7 machine and it created the hex file for me. (after I changed the output setting to generate HEX..)
So for future info..  This won't work on WinXP...

closed 2019-04-21 by jbuszkie