michar71/Open-DSO-150 #34: Scope shows switching to DC/AC but sees to be stuck in GND mode?

michar71, 2019-07-31:
There seems to be sometimes an issue that the scope is stuck in GND mode (No signal) no matter what the switch position is. 
Or maybe this is something else, needs to investigate....

deceusterp, 2019-09-19:
First of all, thank you for the great job you've done on improving the software.
It works even on my 'fake' board!

My scope was also stuck in GND mode and I couldn't get it back to work again.
Then I found that there was no negative voltage anymore on my analogue board.
Apparently the ICL7660 IC that generates the V- voltage was failing.
The ICL7660 IC seems to be prone to latch up and in that case the internal wiring in the chip blows up.
I removed the failing ICL7660 and supplied an external -9Volt to the circuit.
And everything worked again :-)
I have now ordered a new ICL7660 and will test it when received.
To prevent future latch up, one can connect a diode to Vout of ICL7660 on pin5 (anode) and ground (cathode). This will prevent (unintended) positive voltage on pin 5 that can cause latch up.
[image]

b0tm1nd, 2023-03-12:
> First of all, thank you for the great job you've done on improving the software. It works even on my 'fake' board!
>
> My scope was also stuck in GND mode and I couldn't get it back to work again. Then [...]

Hello. Not too sure if this is very on topic, but I have just assembled a fresh new DSO150 and it was working until I connected the red clip to the test terminal as described in instruction. It was working fine, showing an oscilloscope and I tried to play with some buttons / settings. When suddenly the diagram started rising up and something blew up. After restart it would boot up but not show any diagram also settings seem not to change anymore.
After disassembling found out that it was ICL7660 which blown up. 
So trying to find any hints on why this has happened and would replacing the seemingly very inexpensive chip solve it all, so was it just the faulty chip?
I was powering it with 9 volts using the adjustable power supply.