michar71/Open-DSO-150 #41: Sigrok support?

Harvie, 2019-11-24:
Can you please output waveform to sigrok / pulseview software over serial port?
https://sigrok.org/wiki/Supported_hardware

michar71, 2020-06-07:
Interesting idea. I'm not seeing a "Generic Protocol" for sigrok that can be implemented on devices.
Need to figure out what the best approach would be.

Serial port is definitely too slow for real time capture so the data would be buffered in the scope and than transmitted but things like trigger mode could be set up remotely through the UI....

Harvie, 2020-06-07:
For logical analyzers there is something called "SUMP protocol", which is kinda generic protocol used by opensource projects. However there are several downsides and not all implementations are completely compatible, some are tied to capabilities specific hardware (hardcoded sample rates, numer of channels, etc...), so it always takes some tuning because lack of autodetection... You might take a look at it, but there is no support for analog signals in SUMP. Anyway i guess it would be cool to develop some opensource protocol which would be suitable for logic analyzers, scopes and multimeters and use it to connect DSO-150 and sigrok...