michar71/Open-DSO-150 #42: WAVE2 Support?

Harvie, 2019-11-25:
What about this new two channel device?
https://jyetech.com/wave2-2-channel-portable-oscilloscope/

[image]

michar71, 2020-05-05:
Just found that (Because I finally have some time to work through the bug list...).
They split up the Acquisition/UI part from the UI controls by using two processors but it all looks overly complicated... (Schematic is on their webpage...)

With a few simple things they could have improved DSO-150 a lot:
- Connect USB
- Use SPI for Display
- Don't connect Buttons to Display Control Lines
- Add second analog channel

Or go to a STM32F4 ($5 instead of $1.80...). But two processors? Argghh....

(I'll work on DSO150 instead, maybe overclock processor and do a few other improvements....)

closed 2020-05-05 by michar71

Harvie, 2020-05-31:
> But two processors? Argghh....

Do you think you can open KiCAD and implement these improvements? :-)