michar71/Open-DSO-150 #44: Display rotation issues

positron96, 2020-06-07:
Hi. I just flashed my dso and my screen is rotated by 90 degrees. When I tried yesterday's version, my screen was rotated 180 degrees. This is probably caused by recent support of another display.
[image]
[image]

positron96, 2020-06-07:
What helped me was enabling hardcoded ST7789 define and adding additional 180 deg rotation (at least, I think it's rotation)
```
 Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)

diff --git a/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp b/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp
index 5af6f51..b0f2c67 100644
--- a/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp
+++ b/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp
@@ -294,7 +294,10 @@ void setRotation(uint8_t x)
   //perform hardware-specific rotation operations...
    uint16_t t = 0;
  
-  if(isST7789) x^=1; // Landscape & portrait are inverted compared to ILI
+  if(isST7789) {
+    x^=1; // Landscape & portrait are inverted compared to ILI
+    x = (x + 2)%4;
+  }
 
    switch (x)
    {
```

michar71, 2020-06-07:
Hm.... So maybe it's not only the 90 Deg rotation based on controller chip but also 180 Deg rotation based ho they mount it.....

Controller chip ID doesn't seem to work for me I only get back 0. (But register read might be screwed up in general....) As I don't have that display it's hard to test for me...

For the moment I'll use your solution ;-)

positron96, 2020-06-08:
I think there might be a more elegant solution, e.g. x = (x+1)%4 instead of coming and adding.

positron96, 2020-06-09:
Actually, reading chip ID does work for me. That commented out line `int displayId=readReg32(0x04) & 0xffff; //0x9341 for ili9341 ?` gives me 0x8552 for displayID (readReg32 returns 0xFC858552) so the check passes. 0x8552 seems to correspond to ST7789 according to its datasheet. https://www.waveshare.com/w/upload/a/ae/ST7789_Datasheet.pdf

positron96, 2020-06-09:
And my working code for rotation is now this (90 deg rotation replaces XOR and 180 rotation):
```
  if(isST7789) {
    x = (x+1) % 4; // Landscape & portrait are inverted compared to ILI
  }
```

michar71, 2020-06-09:
On my unit it just reads back 0's from the display controller. Might have to hook up a scope to see what's going on on the data bus.
Will also try reading back other registers to see if this is a general issue or just that one register...

luisgongod, 2021-06-14:
Hi, I think  I have the same issue. If I flash the .hex from https://github.com/Shayshez/EuroScope and the Display is 90 degrees (as your first post). Cloned and build in platformio (vscode) the repo from michar71 but after flashing is all blank. Tried to un-comment the lines you mentioned above but nothing shows. 
Is there an extra setting/config I need to do ? could you share the binary for your solution? Thanks!