michar71/Open-DSO-150 #47: Swapped RED and BLUE color channels on the LCD

TheCrypt0, 2020-06-30:
Thank you for the great work with this firmware.
I've just flashed it on my scope and I noticed a weird behavior where the red and blue color channels seems to be swapped. Nothing too fancy and (probably) easily fixable but worth mentioning.

[image]

I took a look at the code and in `global.h` the analog line should actually be yellow:
```c
// display colours
#define AN_SIGNAL1    ILI9341_YELLOW
#define DG_SIGNAL1    ILI9341_RED
#define DG_SIGNAL2    ILI9341_BLUE
#define DG_SIGNAL3    ILI9341_GREEN
```

As expected the `ILI9341_YELLOW` is defined in the Adafruit library:
```c
#define ILI9341_YELLOW      0xFFE0      /* 255, 255,   0 */
```

Maybe a change with the new LCD controller?

gregcrago, 2020-11-02:
I have Original JYETech 150 and I get BLUE text as well. I changed my Stats and Cursor text my changing in display.cpp line 974 tft_setTextColor(ILI9341_BLUE, ILI9341_BLUE);	// should now get RED text and also change line 682 to tft_setTextColor(ILI9341_BLUE, ILI9341_BLUE); // should now get RED text
It would be nice to know all the the actual colors in this unit. Analog is set to YELLOW, but I see CYAN

gregcrago, 2020-11-02:
in global.h I changed:
#define AN_SIGNAL1    ILI9341_CYAN	// now displays YELLOW
#define DG_SIGNAL1    ILI9341_BLUE	// now displays RED

now it looks like the pictures, the BLUE text overlay and digital signal was very hard to read

TheCrypt0, 2020-12-24:
I've finally managed to fix it. Just replace the following lines in the file:

https://github.com/michar71/Open-DSO-150/blob/master/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp#L53
```c
...
writeRegister8(ILI9341_MADCTL , ILI9341_MADCTL_MY | ILI9341_MADCTL_RGB);
...
```

https://github.com/michar71/Open-DSO-150/blob/master/Src/TFTLib/Adafruit_TFTLCD_8bit_STM32.cpp#L308
```c
void setRotation(uint8_t x)
{
  //perform hardware-specific rotation operations...
   uint16_t t = 0;

   if ( dispControllerId==0x8552)
   {
	   x = (x+1) % 4; // Landscape & portrait are inverted compared to ILI
   }

//#ifdef IS_ST7789
//   x = (x+1) % 4; // Landscape & portrait are inverted compared to ILI
//#endif
 

   switch (x)
   {
   case 2:
     t = ILI9341_MADCTL_MX | ILI9341_MADCTL_RGB;
     break;
   case 3:
     t = ILI9341_MADCTL_MV | ILI9341_MADCTL_RGB;
     break;
  case 0:
    t = ILI9341_MADCTL_MY | ILI9341_MADCTL_RGB;
    break;
   case 1:
     t = ILI9341_MADCTL_MX | ILI9341_MADCTL_MY | ILI9341_MADCTL_MV | ILI9341_MADCTL_RGB;
     break;
   }
   writeRegister8(ILI9341_MADCTL, t ); // MADCTL
   // For 9341, init default full-screen address window:
   setAddrWindow(0, 0, TFTWIDTH - 1, TFTHEIGHT - 1); // CS_IDLE happens here
}
```