michar71/Open-DSO-150 #57: Versions after 3.1 won't boot on fake DSO

ms4sman, 2021-02-26:
Somewhere around commit b575ba2649ad0b60a90e034403b93b7e50c128c3 a change was made that makes me unable to use this firmware on my fake DSO. I bought a fake before realizing I needed to be on the lookout, so decided to try this software. The latest commit just gives a white screen, and after wiping my config and resetting the one flash bit mentioned in some other issues, I still just get a white screen. But a much older version I found pre-built worked fine. So I spent last evening building and flashing each version until I found where they stopped working, and the two commits before the one mentioned, the screen is sideways but it does boot. But starting at b575ba2649ad0b60a90e034403b93b7e50c128c3 and on, I just get a white screen.

Any ideas what might have been changed that caused this? If you need to know anything specific about my particular knockoff and its chips or anything, just let me know. In the meantime, I am using commit 08f6c5a2c314170c226403a3f781455e252b8a9e as it is the last one that seems to work for me without the screen rotation issue.

Thanks!

totolook, 2021-07-30:
Hi, commented `#define OVERCLOCK` in `global.h`and rebuild the project seem the fake dso doesn't support the overclock.

AapoTahkola, 2022-04-24:
I am having similar issues. I kind of torched one of the JP1 pins so I guess have to use ST link from now on. TVisual studio debugger misbehaves so badly it is difficult to say what is going on but I got it to blink by turning blink_led to infinite loop. I think it was hanging on first printf. Some folks suggested the problem might be that it is STM32F103C6 not STM32F103C8.
It does say STM32F103C8 and it does not look like it was sanded so dunno.
I think it also hanged on __HAL_FLASH_PREFETCH_BUFFER_ENABLE() which is not available on all chips.
st-flash says it has 64kb flash and which is true for STM32F103C8 not C6.
Device connected is: F1 Medium-density device, id 0x20036410
STM32F103C8T6

I have had this dso 150 few years unused but now I need a logic analyzer with two channels which prompted me to this quest.

Harvie, 2026-09-15:
This patch fixes it for me...
[0001-HARVIE-EDITION.patch](https://github.com/user-attachments/files/32228224/0001-HARVIE-EDITION.patch)

Harvie, 2026-09-15:
it should be even possible to boot on HSI, test if HSE clock is stable enough for overclock and then dynamicaly decide what clock settings we wanna use.